IRTrail/G2E-Filametrix #5: Revo Voron Hotend request

discreteaudio, 2024-02-19:
Would love to build your design to go with my ERCF v2.  Unfortunately, I don't see that as an option.  Any chance to add it to the mix?

IRTrail, 2024-02-19:
Check it ooooouuuuut!

https://github.com/IRTrail/G2E-Filametrix/tree/main/Printheads/Revo

Let me know if there are any issues with it. I just revised the CAD and haven't printed one myself.

closed 2024-02-19 by IRTrail

discreteaudio, 2024-02-19:
I'll give it a go as soon as my current print finishes up!  Looking forward to the build for sure!  Thanks for the rapid response!

discreteaudio, 2024-02-21:
Got it printed last night.  Hoping to put it together this weekend after a trip to the hardware store for misc hardware.  I will keep you posted on progress.  So far everything looks fantastic.

discreteaudio, 2024-02-27:
[image]
[image]
[image]

I had to file out material to allow the arm to have enough travel to cut filament and retract easily.  If you edit the design, take out the black bit.

IRTrail, 2024-02-28:
Thanks for the pictures. I added a recess to help with that. I just pushed it, so give that a try to see what you think.

discreteaudio, 2024-02-29:
That did the trick!  Now to finish building everything.  LoL
Thanks again!