IRTrail/G2E-Filametrix #11: Bambu Hotend Support

barrenechea, 2024-04-02:
Hi! Could support for the Bambu Hotend be added? It would be greatly appreciated!
And thanks in advance for this fantastic repo!!

IRTrail, 2024-04-02:
You might try this:
https://github.com/sorted01/Filametrix/tree/main/STL/Bambu%20Toolhead

I haven't worked on the Bambu yet, but the only difference would be the small pad which acts as a stop for the stylized arm. If it works, cool. If not, try the "Wide Arm" here:
https://github.com/IRTrail/G2E-Filametrix/blob/main/Extruder%20and%20Cutter%20Parts/Twinsor/STL/G2E_Filametrix_Wide%20Arm%20ADXL%20stop.STL

That one uses the top ADXL screw just like the original Filametrix.

I'll try to get this ported when I have some time, but hopefully the above will work in the interim.

alfadex, 2024-08-20:
Hi , for hotend e3d v6, can i use this also ?
https://github.com/sorted01/Filametrix/tree/main/STL/V6%20R6

IRTrail, 2024-08-20:
> Hi , for hotend e3d v6, can i use this also ? https://github.com/sorted01/Filametrix/tree/main/STL/V6%20R6

Yes, that will most likely work. You might need the "wide arm" linked above.

closed 2024-08-20 by IRTrail