IRTrail/G2E-Filametrix #12: Cutting arm travel is limited

WegnerDan, 2024-05-22:
I have a hard time describing this, so I made some videos:

Without the G2E on top, the travel of the cutting arm is fine:

https://github.com/IRTrail/G2E-Filametrix/assets/6908247/3e7bb79e-3903-4199-9b67-89a52a9f2ef2

When the hotend is installed to the toolhead mount, G2E limits the travel of the cutting arm more than it needs to. 

https://github.com/IRTrail/G2E-Filametrix/assets/6908247/6f8d3ab7-c530-49bf-a3ca-5c4a46384f43

I think it makes sense to either extend the top of the cutting arm outward a couple mm (1)
or allow the cutting arm to travel outward a little more (2).

This would make cutting the blade to the correct length a little easier. 
[image]

What do you think?

WegnerDan, 2024-05-22:
Some more information:

The blade in the video was shortened to 25.3 mm, because it was interfering with the filament.

[image]

This is a blade cut to 26 mm:
[image]

[image]

As a test, I've filed down the area marked with 2 in the screenshot above and the filament now clears the 26mm blade

IRTrail, 2024-05-23:
When I made this, I matched the angular deflection of the original version by Sorted. The gap in the space between the arm and the G2E bodies is about half a milimeter, iirc (been a monute).

The stop is set to match deflection as well.

There's also the "wide arm" which retains the original hook that stops against the ADXL screw if you like. It's fully interchangeable.

You're the first to mention this issue. I'll keep an eye out for more reports.

closed 2024-05-23 by IRTrail